kaloz/mwlwifi #396: anybody having issues since OpenWrt v21-rc1 on Linksys WRT 3200ACM/WRT32X with WIFI

p3x-robot, 2021-06-10:
The Wifi works, but for example RDP has re-connecting issues (always disconnect, reconnect, but is not acceptable level), and on Facebook videos sometimes stops and have to go to another video play and then go to the original video an dplay then, that is how it can only work.

with v19 wifi was perfect (though max seed 866 MBit, no 3 x 160mhz streams etc = 3200mbit...)

What is weird is that with Linksys WRT1900ACSv2 it is no RDP/videos streaming issues at all.

BrainSlayer, 2021-06-10:
> The Wifi works, but for example RDP has re-connecting issues (always disconnect, reconnect, but is not acceptable level), and on Facebook videos sometimes stops and have to go to another video play [...]

160 and 80+80 is always 2x2. its basicly 2 separate 2x2 80 mhz channels. thats how its implemented on all chipsets i know.

WildByDesign, 2021-06-15:
> The Wifi works, but for example RDP has re-connecting issues (always disconnect, reconnect, but is not acceptable level), and on Facebook videos sometimes stops and have to go to another video play [...]

Major issues here with WRT3200ACM and 21.02 RC1/RC2 regarding wireless. It seems this is going to be a difficult issue for developers to figure out the cause and likely 21.02 is going to cause a lot of complaints with users of this chipset.

19.07.x series has been stable and great wireless with WRT3200ACM and also DavidC OpenWrt builds have likely been the best up until DSA arrived.

Everything seems to have gone downhill since DSA.

p3x-robot, 2021-07-14:
i think error is an android facebook issue, it is happening with other users specific with facebook android videos, so closing this

closed 2021-07-14 by p3x-robot

moriel5, 2022-06-14:
I have just updated right now, will see what issues will crop up (I currently don't seem to have internet access on 5Ghz, I'll look into that tonight), however on 19.07 I generally had issues with one of my Android devices unable to access the Internet, regardless of the wireless network, regardless of the band, at random (and rare) intervals, which usually alerted itself after a week or two.