kaloz/mwlwifi #400: kernel panics with OpenWRT 21.x snapshots

ByteEnable, 2021-08-30:
I can at will kernel panic 21.x snapshots I am running.  I have been trying to debug my 5G wifi issues and ran into this issue only on the wireless side.

Wireless config:

`config wifi-device 'radio0'
        option type 'mac80211'
        option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option country 'US'
        option cell_density '0'
        option log_level '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option macaddr '30:23:03:df:49:2a'
        option disassoc_low_ack '0'
        option encryption 'psk2+ccmp'
        option key 'toomuchFUN'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option disabled '1'
        option country 'US'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
        option macaddr '30:23:03:df:49:29'

config wifi-device 'radio2'
        option type 'mac80211'
        option path 'platform/soc/soc:internal-regs/f10d8000.sdhci/mmc_host/mmc0/mmc0:0001/mmc0:0001:1'
        option channel '34'
        option band '5g'
        option htmode 'VHT80'
        option disabled '1'

config wifi-iface 'default_radio2'
        option device 'radio2'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

`

Steps to reproduce via ssh to the router:
```
iw wlan0 interface add mon0 type monitor
ifconfig mon0 up
tcpdump -vvvi mon0 ether host xx:xx:xx:xx:xx:xx
```
Open another ssh session to the router :
`iperf3 -s`

From another wireless device launch iperf3 as a client. I am using magic perf from the google playstore on my android phone. Will kernel panic usually in about 20 seconds.

Linux OpenWrt 5.4.143 #0 SMP Sun Aug 29 22:37:47 2021 armv7l GNU/Linux
[... 152 lines cut ...]
Aug 28 20:22:53 OpenWrt kernel: [  671.740464] fea0: c107fed8 c06befc4 dfb62d70 dfb62cc0 0000012c c0b4fcc0 c107fed0 c06bf2a0
Aug 28 20:22:53 OpenWrt kernel: [  671.748678] fec0: c0c04f58 00009111 00000000 000e6b78 c107fed0 c107fed0 c107fed8 c107fed8
Aug 28 20:22:53 OpenWrt kernel: [  671.756892] fee0: c107ff08 00000004 00000003 c0c0308c c0c03080 40000003 c107e000 00000100
Aug 28 20:22:53 OpenWrt kernel: [  671.765105] ff00: c107ff08 c01012ec cb92d600 c104d800 cb92d600 c0c03080 c0b4c220 00000008
Aug 28 20:22:53 OpenWrt kernel: [  671.773319] ff20: c0b4c1ac c0b4ef00 c0b4ef00 00009110 c0c03d00 04208040 c107e000 00000000
Aug 28 20:22:53 OpenWrt kernel: [  671.781534] ff40: 00000000 c101ccc0 c107e000 c0c0cf24 ffffe000 c1037e64 c101cda4 c012c018
Aug 28 20:22:53 OpenWrt kernel: [  671.789748] ff60: 00000000 c014ad74 c101cd80 c101cd40 00000000 c107e000 c014aca4 c101ccc0
Aug 28 20:22:53 OpenWrt kernel: [  671.797962] ff80: c1037e64 c0145b34 00000001 c101cd40 c01459ec 00000000 00000000 00000000
Aug 28 20:22:53 OpenWrt kernel: [  671.806176] ffa0: 00000000 00000000 00000000 c0100148 00000000 00000000 00000000 00000000
Aug 28 20:22:53 OpenWrt kernel: [  671.814390] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Aug 28 20:22:53 OpenWrt kernel: [  671.822603] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
Aug 28 20:22:53 OpenWrt kernel: [  671.830820] [<c06bf808>] (process_backlog) from [<c06befc4>] (__napi_poll+0x28/0x134)
Aug 28 20:22:53 OpenWrt kernel: [  671.838685] [<c06befc4>] (__napi_poll) from [<c06bf2a0>] (net_rx_action+0xc8/0x24c)
Aug 28 20:22:53 OpenWrt kernel: [  671.846376] [<c06bf2a0>] (net_rx_action) from [<c01012ec>] (__do_softirq+0x104/0x29c)
Aug 28 20:22:53 OpenWrt kernel: [  671.854247] [<c01012ec>] (__do_softirq) from [<c012c018>] (run_ksoftirqd+0x2c/0x38)
Aug 28 20:22:53 OpenWrt kernel: [  671.861944] [<c012c018>] (run_ksoftirqd) from [<c014ad74>] (smpboot_thread_fn+0xd0/0x18c)
Aug 28 20:22:53 OpenWrt kernel: [  671.870158] [<c014ad74>] (smpboot_thread_fn) from [<c0145b34>] (kthread+0x148/0x14c)
Aug 28 20:22:53 OpenWrt kernel: [  671.877936] [<c0145b34>] (kthread) from [<c0100148>] (ret_from_fork+0x14/0x2c)
Aug 28 20:22:53 OpenWrt kernel: [  671.885189] Exception stack(0xc107ffb0 to 0xc107fff8)
Aug 28 20:22:53 OpenWrt kernel: [  671.890261] ffa0:                                     00000000 00000000 00000000 00000000
Aug 28 20:22:53 OpenWrt kernel: [  671.898474] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Aug 28 20:22:53 OpenWrt kernel: [  671.906687] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
Aug 28 20:22:53 OpenWrt kernel: [  671.913331] Code: e50430a0 e1c020d0 e5801000 e5801004 (e5823004)
Aug 28 20:22:53 OpenWrt kernel: [  671.919509] ---[ end trace 819e11397d0d8fc2 ]---
`

BrainSlayer, 2021-08-31:
maybe you should just avoid to use the monitor mode, since this mode isnt supported by the driver. i just wonder why you are able to use monitor mode since the driver does not offer the required support flags

ByteEnable, 2021-08-31:
Literally states in the driver readme how to debug:

Monitor interface for debug

    Create moinitor interface mon0:

    iw wlan0/wlan1 interface add mon0 type monitor
    ifconfig mon0 up

    Use tcpdump to dump dhcp packets:

    tcpdump -vvvi mon0 -n port 67 and port 68

    Use tcpdump to dump icmp packets:

    tcpdump -vvvi mon0 icmp

ByteEnable, 2021-09-04:
Able to induce kernel panic on 19.07.8 as well.

`Sep  3 22:25:27 OpenWrt logread[1634]: Logread connected to 192.168.50.83:5555
Sep  3 22:26:31 OpenWrt dropbear[4527]: Child connection from 192.168.50.83:46458
Sep  3 22:26:35 OpenWrt dropbear[4527]: Password auth succeeded for 'root' from 192.168.50.83:46458
Sep  3 22:26:35 OpenWrt dropbear[4527]: Exit (root): Disconnect received
Sep  3 22:27:45 OpenWrt mac80211: Failed command: iw phy phy0 set antenna 0xffffffff 0xffffffff
Sep  3 22:27:45 OpenWrt netifd: radio0 (4765): command failed: Not supported (-95)
Sep  3 22:27:45 OpenWrt mac80211: Failed command: iw phy phy0 set distance 0
Sep  3 22:27:45 OpenWrt hostapd: Configuration file: /var/run/hostapd-phy0.conf
Sep  3 22:27:45 OpenWrt kernel: [  635.257672] ieee80211 phy0: change: 0xffffffff
Sep  3 22:27:45 OpenWrt kernel: [  635.362941] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Sep  3 22:27:45 OpenWrt kernel: [  635.370374] br-lan: port 2(wlan0) entered blocking state
Sep  3 22:27:45 OpenWrt kernel: [  635.375731] br-lan: port 2(wlan0) entered disabled state
Sep  3 22:27:45 OpenWrt kernel: [  635.381172] device wlan0 entered promiscuous mode
Sep  3 22:27:45 OpenWrt hostapd: wlan0: interface state UNINITIALIZED->HT_SCAN
Sep  3 22:27:45 OpenWrt kernel: [  635.387151] ieee80211 phy0: change: 0x100
Sep  3 22:27:45 OpenWrt kernel: [  635.395859] ieee80211 phy0: change: 0x40
Sep  3 22:27:46 OpenWrt kernel: [  635.610801] ieee80211 phy0: change: 0x40
Sep  3 22:27:46 OpenWrt kernel: [  635.820877] ieee80211 phy0: change: 0x40
Sep  3 22:27:46 OpenWrt kernel: [  635.916937] ieee80211 phy0: change: 0x100
Sep  3 22:27:46 OpenWrt hostapd: Using interface wlan0 with hwaddr 30:23:03:df:49:2a and ssid "OpenWrt"
Sep  3 22:27:46 OpenWrt kernel: [  635.968897] ieee80211 phy0: change: 0x100
Sep  3 22:27:46 OpenWrt kernel: [  635.977922] ieee80211 phy0: change: 0x42
Sep  3 22:27:46 OpenWrt kernel: [  636.146019] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Sep  3 22:27:46 OpenWrt kernel: [  636.152474] br-lan: port 2(wlan0) entered blocking state
Sep  3 22:27:46 OpenWrt kernel: [  636.157811] br-lan: port 2(wlan0) entered forwarding state
Sep  3 22:27:46 OpenWrt hostapd: wlan0: interface state HT_SCAN->ENABLED
Sep  3 22:27:46 OpenWrt hostapd: wlan0: AP-ENABLED
Sep  3 22:27:46 OpenWrt netifd: Network device 'wlan0' link is up
Sep  3 22:28:26 OpenWrt kernel: [  676.323211] ieee80211 phy0: change: 0x8
Sep  3 22:28:59 OpenWrt dropbear[5307]: Child connection from 192.168.50.83:46460
Sep  3 22:29:00 OpenWrt odhcpd[1859]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sep  3 22:29:03 OpenWrt dropbear[5307]: Password auth succeeded for 'root' from 192.168.50.83:46460
Sep  3 22:29:25 OpenWrt kernel: [  735.206045] device mon0 entered promiscuous mode
Sep  3 22:29:59 OpenWrt hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
Sep  3 22:30:00 OpenWrt hostapd: wlan0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx
Sep  3 22:30:00 OpenWrt hostapd: wlan0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sep  3 22:30:01 OpenWrt odhcpd[1859]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sep  3 22:30:01 OpenWrt hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Sep  3 22:30:03 OpenWrt dnsmasq-dhcp[2697]: DHCPDISCOVER(br-lan) xx:xx:xx:xx:xx:xx
Sep  3 22:30:03 OpenWrt dnsmasq-dhcp[2697]: DHCPOFFER(br-lan) 192.168.1.225 xx:xx:xx:xx:xx:xx
Sep  3 22:30:03 OpenWrt dnsmasq-dhcp[2697]: DHCPDISCOVER(br-lan) xx:xx:xx:xx:xx:xx
Sep  3 22:30:03 OpenWrt dnsmasq-dhcp[2697]: DHCPOFFER(br-lan) 192.168.1.225 xx:xx:xx:xx:xx:xx
Sep  3 22:30:03 OpenWrt dnsmasq-dhcp[2697]: DHCPDISCOVER(br-lan) xx:xx:xx:xx:xx:xx
Sep  3 22:30:03 OpenWrt dnsmasq-dhcp[2697]: DHCPOFFER(br-lan) 192.168.1.225 xx:xx:xx:xx:xx:xx
Sep  3 22:30:03 OpenWrt dnsmasq-dhcp[2697]: DHCPREQUEST(br-lan) 192.168.1.225 xx:xx:xx:xx:xx:xx
Sep  3 22:30:03 OpenWrt dnsmasq-dhcp[2697]: DHCPACK(br-lan) 192.168.1.225 xx:xx:xx:xx:xx:xx Rob-s-S10e
Sep  3 22:30:05 OpenWrt odhcpd[1859]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sep  3 22:30:44 OpenWrt kernel: [  814.028018] Unable to handle kernel NULL pointer dereference at virtual address 00000004
Sep  3 22:30:44 OpenWrt kernel: [  814.036170] pgd = c0004000
Sep  3 22:30:44 OpenWrt kernel: [  814.038888] [00000004] *pgd=00000000
Sep  3 22:30:44 OpenWrt kernel: [  814.042483] Internal error: Oops: 817 [#1] SMP ARM
Sep  3 22:30:44 OpenWrt kernel: [  814.047294] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc rfcomm nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack mwifiex_sdio mwifiex iptable_mangle iptable_filter ip_tables hidp hci_uart crc_ccitt btusb btmrvl_sdio btmrvl btintel bnep bluetooth hid evdev input_core mwlwifi mac80211 cfg80211 compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ecdh_generic kpp ecb cmac gpio_button_hotplug
[... cut ...]
Sep  3 22:30:44 OpenWrt kernel: [  814.291178] Function entered at [<c0101628>] from [<c012d0d8>]
Sep  3 22:30:44 OpenWrt kernel: [  814.297035] Function entered at [<c012d0d8>] from [<c0147664>]
Sep  3 22:30:44 OpenWrt kernel: [  814.302892] Function entered at [<c0147664>] from [<c0144740>]
Sep  3 22:30:44 OpenWrt kernel: [  814.308749] Function entered at [<c0144740>] from [<c01075c8>]
Sep  3 22:30:44 OpenWrt kernel: [  814.314607] Code: e50430a0 e1c020d0 e5806000 e5806004 (e5823004)
Sep  3 22:30:44 OpenWrt kernel: [  814.320736] ---[ end trace 18bb1355718c00ff ]---`

BrainSlayer, 2021-09-04:
i just can tell you that i find no evidence that the driver has monitor mode support.

Izaic, 2021-09-05:
I'm finding lockups with 21. as well... Where before in 19.07.8 I was able to run 5GHz without too many issues, after upgrading I now have constant stability problems.

BrainSlayer, 2021-09-05:
there where always stability issues, especially if using IOS / apple devices.