kaloz/mwlwifi #404: Strange connection lose on WRT1200ac v1

thekk1, 2021-10-31:
Hi folks, I stumbled over a strange behavior:

**Setup:**
- Linksys WRT1200ac v1
- OpenWrt 21.02.0 r16279-5cc0535800
- Samsung S21 4G
- Laptop with Intel Wi-Fi 6 AX200
- Old mobile phone Huawei Honor 6
- Test router Cudy WR1300

**Configs:**
<details>
<summary>network</summary>

```

config interface 'loopback'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'
	option device 'lo'

config globals 'globals'
	option ula_prefix 'fdea:b0b8:6814::/48'

config interface 'lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ipaddr '192.168.10.1'
	option device 'br-lan'

config device 'lan_eth0_1_dev'
	option name 'eth0.1'
	option macaddr 'b4:4b:d6:XX:XX:XX'

config interface 'wan'
	option proto 'dhcp'
	option peerdns '0'
	option metric '1'
	option device 'wan'

config device 'wan_eth0_2_dev'
	option name 'eth0.2'
	option macaddr 'b4:4b:d6:XX:XX:XX'

config interface 'wan6'
	option proto 'dhcpv6'
	option reqprefix 'auto'
	option device 'wan'
	option auto '0'
	option reqaddress 'none'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '0 1 2 3'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '4'

config interface 'mwan'
	option proto 'dhcp'
[... 271 lines cut ...]
[21417.604875] wlan1: send auth to ba:8c:6f:84:e3:58 (try 1/3)
[21417.614341] wlan1: authenticated
[21417.626832] ieee80211 phy1: change: 0x2
[21417.631083] br-lan: port 7(wlan1-1) entered disabled state
[21417.642762] wlan1: associate with ba:8c:6f:84:e3:58 (try 1/3)
[21417.650874] wlan1: RX AssocResp from ba:8c:6f:84:e3:58 (capab=0x421 status=0 aid=1)
[21417.741838] ieee80211 phy1: change: 0x2
[21417.745711] wlan1: associated
[21417.821694] br-lan: port 7(wlan1-1) entered blocking state
[21417.827235] br-lan: port 7(wlan1-1) entered forwarding state
```
</details>

**Tries to analyse and solve:**
- Using an other router (Cudy WR1300) with same setup -> Works
- Using a wifi stick (TL-WN722N) with Linksys and connect over them -> Works
- Enable/disable some settings on router (2.4 & 5 GHz):
1. Disassociate On Low Acknowledgement -> No improvement
2. Short Gi -> No improvement
- Enable/disable some settings on mobile phone:
1. Detect suspicious networks -> No improvement
2. WiFi energy saving mode -> No improvements

**Conclusion:**
It seems to be a problem between on chipsets or driver settings of the Marvell 88W8864 but I am not able make a deeper investigation. Maybe someone can help me out?

stamak, 2021-11-29:
I am experiencing the same issue with my Linksys WRT1200AC after upgrading to `OpenWrt 21.02.1 r16325-88151b8303` 5Ghz WiFi constantly does not work (no Internet, ping `Request timeout for icmp_seq 0`) from time to time for ~1 minutes on devices.
Initially I thought that it was due to `WPA3` and `802.11w` but no, switching to `WPA2` and `disabling` of 802.11w does not help.

During issues In logs I only see such messages
```
Mon Nov 29 20:31:18 2021 kern.debug kernel: [80963.169051] ieee80211 phy0: Stop BA xx:xx:32:b3:xx:0d
Mon Nov 29 20:31:23 2021 kern.debug kernel: [80968.151710] ieee80211 phy0: Mac80211 start BA xx:xx:32:b3:xx:0d
```

Enabled logging for `hostapd` and there no warning/errors and any suspicious messages

driver info
```
cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.8.0-20181210
firmware version: 0x0702091a
power table loaded from dts: yes
firmware region code: 0x0
mac address: 00:25:9c:xx:xx:3e
2g: disable
5g: enable
antenna: 2 2
irq number: 73
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: 0de90e6b
iobase1: 00b85884
tx limit: 768
rx limit: 64
qe trigger number: 3571382
```

```
 modinfo mwlwifi
module:		/lib/modules/5.4.154/mwlwifi.ko
license:	GPL v2
depends:	mac80211,cfg80211
name:		mwlwifi
vermagic:	5.4.154 SMP mod_unload ARMv7 p2v8
```

Seems support for this driver is not provided anymore as for 29days no activities.
I would help with debugging of this issue but do not know how I can enable it for kernel module.
If case nobody helps only downgrade to 19.07.x is left.

arinc9, 2021-11-30:
The wireless cut-out issue is fixed on OpenWrt and a patch is submitted for Linux upstream.

https://forum.openwrt.org/t/users-needed-to-test-wi-fi-stability-on-linksys-wrt3200acm-wrt32x-on-openwrt-21-02/101700?u=arinc9

@stamak

stamak, 2021-11-30:
@arinc9  thx, its cool.

Can I safely test it with https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=mvebu%2Fcortexa9&id=linksys_wrt1200ac image?

arinc9, 2021-11-30:
> Can I safely test it with https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=mvebu%2Fcortexa9&id=linksys_wrt1200ac image?

Yup.

leocrawford, 2021-11-30:
I found with an honor view 10 that I had to reduce beacon interval from 100 down to 75, otherwise the phone sent out ARP requests but never saw the response.

intgr, 2021-12-01:
Alternatively one can download the package from https://github.com/adworacz/openwrt-linksys-wireless/tree/patched-mac80211/releases/21.02.1-patch-mac80211-5.10

And install it with `opkg`:

```shell
opkg install ./kmod-mac80211_5.4.154\%2B5.10.68-1-1_arm_cortex-a9_vfpv3-d16.ipk --force-reinstall
```

You can verify that it's installed by checking the dates of kernel modules
```
root@OpenWrt:~# ls -la /lib/modules/5.4.154/ |grep -C3 mac80211
-rw-r--r--    1 root     root          4380 Oct 24 09:01 iptable_filter.ko
-rw-r--r--    1 root     root          4308 Oct 24 09:01 iptable_mangle.ko
-rw-r--r--    1 root     root          4216 Oct 24 09:01 iptable_nat.ko
-rw-r--r--    1 root     root        625000 Sep 22 20:36 mac80211.ko        <-- modification time different from other modules
-rw-r--r--    1 root     root        191224 Oct 24 09:01 mwlwifi.ko
-rw-r--r--    1 root     root        103760 Oct 24 09:01 nf_conntrack.ko
-rw-r--r--    1 root     root          4272 Oct 24 09:01 nf_defrag_ipv4.ko
```

regulskimichal, 2021-12-01:
Yesterday I installed a snapshot and I can see a huge improvement in the stability of 5ghz WLAN. Also, it works with WPA2/WPA3 mixed with IEEE 802.11w set as optional.

regulskimichal, 2021-12-01:
It seems I praised the day before sunset. Still some issues occurring, it is hard to tell from my perspective what is going on but after some time the quality of 5GHz drops, and it's unusable.

stamak, 2021-12-02:
So after 2 days running (OpenWrt SNAPSHOT, `r18244-101300b842`) on my Linksys WRT1200ac - issue has gone, 5GHz WiFi is stable enough, even with configured WPA3/WPA3 mixed and 802.11w optional.
@arinc9 thx and big thx to guys for fixing it!

Ken1Nil, 2021-12-02:
@stamak 
Sounds promising to get 5GHz WiFi working for more versions than WRT1200ac. Like WRT1900ac V1, V2 etc.

Don't know if Snapshots disappear when new are uploaded. Seems like Snapshot is upgraded to `r18254-25bc66eb40` now. 
[https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/](https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/) 
versioninfo: [version.buildinfo](https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/version.buildinfo)

Any specific WiFi settings you used to get it working? Like Channel 36 (non DFS)? What WPA3 settings? 

Did you run the `...initramfs-kernel.bin` aswell? `squashfs-factory` or `squashfs-sysupgrade`?

thekk1, 2021-12-06:
After a few Days of testing the combination of https://forum.openwrt.org/t/users-needed-to-test-wi-fi-stability-on-linksys-wrt3200acm-wrt32x-on-openwrt-21-02/101700?u=arinc9 and this mention:
> I found with an honor view 10 that I had to reduce beacon interval from 100 down to 75, otherwise the phone sent out ARP requests but never saw the response.

closed 2021-12-06 by thekk1