kaloz/mwlwifi #405: Do we support manually setting MCS?

mengzili, 2021-11-29:
Hi all,

I'm using the mwlwifi driver on my Linksys WRT32X router. I'd like to manually set the MCS to certain index for experimental uses. For example, when I tried to set MCS with:
```
iw dev wlan1 set bitrates ht-mcs-2.4 1
```
, it returns with:
```
command failed: Not supported (-95)
```
Does the driver support setting the MCS or is it an issue of `iw`? Alternatively, is it possible to change the driver of WRT32X wireless NIC to others so that I could manually set the MCS myself?

Thanks a lot!

rmandrad, 2021-12-27:
"iw phy1 info" doesn't show setting bitrates as supported ... try on yours "iw phy1 info" and check the supported commands section of the output

mengzili, 2021-12-30:
> "iw phy1 info" doesn't show setting bitrates as supported ... try on yours "iw phy1 info" and check the supported commands section of the output

Thanks rmandrad! Indeed it does not support. I've switched to ath9k-based routers.