kaloz/mwlwifi #406: `radio1` gets disabled when there's two enabled SSID on it

NatoBoram, 2022-04-06:
I have 2 SSID configured on my `radio1`, but I can only get one to work.

Only first enabled:
[image]
Both enabled:
[image]
Only second enabled:
[image]

```yaml
OS: OpenWrt SNAPSHOT armv7l 
Host: Linksys WRT3200ACM 
Kernel: 5.10.109 
Uptime: 34 mins 
Packages: 208 (opkg) 
Shell: ash 
Terminal: /dev/pts/0 
CPU: Marvell Armada 380/385 (Device Tree) (2) 
Memory: 226MiB / 496MiB 
```

This is my configuration for the Wi-Fi

```bash
uci set wireless.default_radio0.key='***********'
uci set wireless.default_radio0.ssid='Nato'
uci set wireless.default_radio0.encryption='psk2'

uci set wireless.default_radio1.key='***********'
uci set wireless.default_radio1.ssid='Nato'
uci set wireless.default_radio1.encryption='psk2'

uci set wireless.default_radio2.key='***********'
uci set wireless.default_radio2.ssid='Nato'
uci set wireless.default_radio2.encryption='psk2'

uci set wireless.wifinet3=wifi-iface
uci set wireless.wifinet3.ssid='Free Wi-Fi'
uci set wireless.wifinet3.encryption='none'
uci set wireless.wifinet3.device='radio1'
uci set wireless.wifinet3.mode='ap'
uci set wireless.wifinet3.network='lan'
uci set wireless.wifinet3.isolate='1'

uci set wireless.radio0.channel='auto'
uci set wireless.radio1.channel='auto'
uci set wireless.radio2.channel='auto'
```

Please let me know if you need more information and how I can get it!

anomeome, 2022-04-06:
[see issue](https://github.com/openwrt/openwrt/issues/9343#issuecomment-1052410915)

NatoBoram, 2022-04-06:
Thanks!

With this solution:

```ash
uci del wireless.default_radio0.macaddr
uci del wireless.default_radio1.macaddr
uci commit wireless
```

I get the expected result:

[image]

closed 2022-04-06 by NatoBoram