kaloz/mwlwifi #409: mwlwifi: version 10.3.9.0-20230311 disconnection issue

pesa1234, 2023-04-20:
@jbsky With latest patch I have a lot of diconnection on both 2.4 and 5ghz radio.(I'm not the only one)
I have WRT1900ACS with 88w8864

olimpiumarius, 2023-04-22:
Sat Apr 22 13:20:07 2023 daemon.info hostapd: phy0-ap0: STA :34:ac:bc WPA: group key handshake completed (RSN)
Sat Apr 22 13:20:07 2023 daemon.info hostapd: phy0-ap0: STA :b0:96 WPA: group key handshake completed (RSN)
Sat Apr 22 13:20:08 2023 daemon.info hostapd: phy0-ap0: STA :80:99 WPA: group key handshake completed (RSN)
Sat Apr 22 13:30:07 2023 daemon.info hostapd: phy0-ap0: STA :34:ac:bc WPA: group key handshake completed (RSN)
Sat Apr 22 13:30:07 2023 daemon.info hostapd: phy0-ap0: STA :80:99 WPA: group key handshake completed (RSN)
Sat Apr 22 13:30:08 2023 daemon.info hostapd: phy0-ap0: STA :b0:96 WPA: group key handshake completed (RSN)
Sat Apr 22 13:40:07 2023 daemon.info hostapd: phy0-ap0: STA :34:ac:bc WPA: group key handshake completed (RSN)
Sat Apr 22 13:40:07 2023 daemon.info hostapd: phy0-ap0: STA :80:99 WPA: group key handshake completed (RSN)
Sat Apr 22 13:40:08 2023 daemon.info hostapd: phy0-ap0: STA :b0:96 WPA: group key handshake completed (RSN)
Sat Apr 22 13:50:07 2023 daemon.info hostapd: phy0-ap0: STA :34:ac:bc WPA: group key handshake completed (RSN)
Sat Apr 22 13:50:07 2023 daemon.info hostapd: phy0-ap0: STA :b0:96 WPA: group key handshake completed (RSN)
Sat Apr 22 13:50:08 2023 daemon.info hostapd: phy0-ap0: STA :80:99 WPA: group key handshake completed (RSN)
Sat Apr 22 14:00:07 2023 daemon.info hostapd: phy0-ap0: STA :34:ac:bc WPA: group key handshake completed (RSN)
Sat Apr 22 14:00:08 2023 daemon.info hostapd: phy1-ap0: STA :8d:3c:49 IEEE 802.11: authenticated
Sat Apr 22 14:00:08 2023 daemon.info hostapd: phy0-ap0: STA :80:99 WPA: group key handshake completed (RSN)
Sat Apr 22 14:00:08 2023 daemon.info hostapd: phy1-ap0: STA :8d:3c:49 IEEE 802.11: associated (aid 1)
Sat Apr 22 14:00:08 2023 daemon.info hostapd: phy0-ap0: STA :b0:96 WPA: group key handshake completed (RSN)
Sat Apr 22 14:00:08 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED :8d:3c:49 auth_alg=open
Sat Apr 22 14:00:08 2023 daemon.info hostapd: phy1-ap0: STA :8d:3c:49 WPA: pairwise key handshake completed (RSN)
Sat Apr 22 14:00:08 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED :8d:3c:49
Sat Apr 22 14:00:10 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) :8d:3c:49
Sat Apr 22 14:00:10 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.15.235 :8d:3c:49
Sat Apr 22 14:00:12 2023 daemon.info hostapd: phy1-ap0: STA :8d:3c:49 WPA: group key handshake completed (RSN)
Sat Apr 22 14:00:15 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) :8d:3c:49
Sat Apr 22 14:00:15 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.15.235 :8d:3c:49
Sat Apr 22 14:00:15 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.15.235 :8d:3c:49
Sat Apr 22 14:00:15 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.15.235 :8d:3c:49
Sat Apr 22 14:00:17 2023 kern.debug kernel: [90551.047328] ieee80211 phy1: Mac80211 start BA :8d:3c:49
Sat Apr 22 14:10:07 2023 daemon.info hostapd: phy0-ap0: STA :34:ac:bc WPA: group key handshake completed (RSN)
Sat Apr 22 14:10:08 2023 daemon.info hostapd: phy0-ap0: STA :80:99 WPA: group key handshake completed (RSN)
Sat Apr 22 14:10:08 2023 daemon.info hostapd: phy0-ap0: STA :b0:96 WPA: group key handshake completed (RSN)
Sat Apr 22 14:10:12 2023 daemon.info hostapd: phy1-ap0: STA :8d:3c:49 WPA: group key handshake completed (RSN)
Sat Apr 22 14:20:08 2023 daemon.info hostapd: phy0-ap0: STA :34:ac:bc WPA: group key handshake completed (RSN)
Sat Apr 22 14:20:08 2023 daemon.info hostapd: phy0-ap0: STA :80:99 WPA: group key handshake completed (RSN)
Sat Apr 22 14:20:08 2023 daemon.info hostapd: phy0-ap0: STA :b0:96 WPA: group key handshake completed (RSN)
Sat Apr 22 14:20:12 2023 daemon.info hostapd: phy1-ap0: STA :8d:3c:49 WPA: group key handshake completed (RSN)
Sat Apr 22 14:23:36 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sat Apr 22 14:28:33 2023 daemon.err uhttpd[2793]: [info] luci: accepted login on / for root from 192.168.15.153
Sat Apr 22 14:30:07 2023 daemon.info hostapd: phy0-ap0: STA :b0:96 WPA: group key handshake completed (RSN)
Sat Apr 22 14:30:08 2023 daemon.info hostapd: phy0-ap0: STA :34:ac:bc WPA: group key handshake completed (RSN)
Sat Apr 22 14:30:08 2023 daemon.info hostapd: phy0-ap0: STA :80:99 WPA: group key handshake completed (RSN)
Sat Apr 22 14:30:12 2023 daemon.info hostapd: phy1-ap0: STA :8d:3c:49 WPA: group key handshake completed (RSN)
Sat Apr 22 14:37:49 2023 daemon.info hostapd: phy1-ap0: STA :3c:d4 IEEE 802.11: authenticated
Sat Apr 22 14:37:49 2023 daemon.info hostapd: phy1-ap0: STA :3c:d4 IEEE 802.11: associated (aid 2)
Sat Apr 22 14:37:49 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED :3c:d4 auth_alg=open
Sat Apr 22 14:37:49 2023 daemon.info hostapd: phy1-ap0: STA :3c:d4 WPA: pairwise key handshake completed (RSN)
Sat Apr 22 14:37:49 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED :3c:d4
Sat Apr 22 14:37:50 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) :3c:d4
Sat Apr 22 14:37:50 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.15.148 :3c:d4
Sat Apr 22 14:37:50 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.15.148 :3c:d4
Sat Apr 22 14:37:50 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.15.148 :3c:d4 Galaxy-J5-2016
Sat Apr 22 14:37:51 2023 daemon.info hostapd: phy0-ap0: STA :fa:37 IEEE 802.11: authenticated
Sat Apr 22 14:37:51 2023 daemon.info hostapd: phy0-ap0: STA :fa:37 IEEE 802.11: associated (aid 4)
[... cut ...]
Sat Apr 22 15:20:08 2023 daemon.info hostapd: phy0-ap0: STA :34:ac:bc WPA: group key handshake completed (RSN)
Sat Apr 22 15:20:08 2023 daemon.info hostapd: phy0-ap0: STA :80:99 WPA: group key handshake completed (RSN)
Sat Apr 22 15:20:08 2023 daemon.info hostapd: phy0-ap0: STA :fa:37 WPA: group key handshake completed (RSN)
Sat Apr 22 15:20:08 2023 daemon.info hostapd: phy0-ap0: STA :b0:96 WPA: group key handshake completed (RSN)
Sat Apr 22 15:20:13 2023 daemon.info hostapd: phy1-ap0: STA :3c:d4 WPA: group key handshake completed (RSN)

pesa1234, 2023-04-22:
@olimpiumarius 

Are you able to compile from source? I think I found the issue...

trinidude4, 2023-04-22:
@pesa1234 Do you think just https://github.com/kaloz/mwlwifi/commit/9f991fdee60e7b251177d911b6def2a77c1259c9 needs to be removed?  If so, I can try to test that out for WRT32X at least.

jbsky, 2023-04-22:
Hello,

First, please confirm that this commit is the source of the problem.

Then, before making a modification, we must take into consideration why this commit was made.

What would be nice is to better describe your problem and how I can reproduce?
(ex conf file of hostapd + wpa_supplicant, openwrt/debian ?)
What jumps out at me in your logs :
```auth_alg=open.```

It looks like the crypto part of your hostapd configuration has dropped.

For this commit, my test case is the following:
_With a network card in station mode, I want it to switch from an encrypted connection to an unencrypted connection._

To summarize, we must take into consideration your test case and mine in order to validate the change.

pesa1234, 2023-04-22:
> @pesa1234 Do you think just [9f991fd](https://github.com/kaloz/mwlwifi/commit/9f991fdee60e7b251177d911b6def2a77c1259c9) needs to be removed? If so, I can try to test that out for WRT32X at least.

In my case works perfectly with that change...

pesa1234, 2023-04-22:
> Hello,
>
> First, please confirm that this commit is the source of the problem.
>
> Then, before making a modification, we must take into consideration why this commit was made.
>
> What would be nice is to [...]

Yes, in my case that commit give me issues.
I'm on openwrt, kernel 5.15, WPA2, the disconnection happen more frequently when there are more than 5-6 devices or when the connection is stressed (example during an iperf3 test)

olimpiumarius, 2023-04-22:
> @olimpiumarius
> 
> Are you able to compile from source? I think I found the issue...

If I had an image I could test it!

pesa1234, 2023-04-22:
> > @olimpiumarius
> > Are you able to compile from source? I think I found the issue...
> 
> If I had an image I could test it!

If you tell me your device and kernel version I can give you compiled `mwlwifi-firmware `and` kmod-mwlwifi`

olimpiumarius, 2023-04-22:
Wrt1900acs v2 SNAPSHOT r22599-1416b9bbe9 kernel 5.15. 107

pesa1234, 2023-04-22:
> Wrt1900acs v2 SNAPSHOT r22599-1416b9bbe9 kernel 5.15. 107

[https://drive.google.com/drive/folders/1jG8Wza7KxSullVDMhpy5Iefkq5AMZ_kM?usp=share_link](url)

olimpiumarius, 2023-04-22:
OpenWrt SNAPSHOT, r22599-1416b9bbe9
root@OpenWrt:~# opkg install /tmp/*.ipk
Package kmod-mwlwifi (5.15.107+2023-03-11-c916ac13-1) installed in root is up to                    date.
Package mwlwifi-firmware-88w8864 (2023-04-21-ebf38570-7) installed in root is up                    to date.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.107-1-65a719                   adf77accc904bee4e1a7b5152e) for kmod-mwlwifi

pesa1234, 2023-04-22:
> OpenWrt SNAPSHOT, r22599-1416b9bbe9 root@OpenWrt:~# opkg install /tmp/*.ipk Package kmod-mwlwifi (5.15.107+2023-03-11-c916ac13-1) installed in root is up to date. Package mwlwifi-firmware-88w8864 [...]

Ops... I compile by myself... So, you can not use (different version I think...) I give to you my Image... 5.15.108 with ksmbd - openvpn -wireguard if you want...

olimpiumarius, 2023-04-22:
I'll try

anomeome, 2023-04-22:
What SSL library in-play.

pesa1234, 2023-04-22:
> What SSL library in-play.

I'm a newbie, sorry, I don't understand

anomeome, 2023-04-22:
What are you running for SSL, _mbedtls_, _wolfssl_, _openssl_:

`opkg list | grep -i ssl`

If unsure I assume the default, maybe try an _openssl_ image. By way of a quick test, I use _openssl_ in my images, you can find the link to my google drive drop off my github.

pesa1234, 2023-04-22:
`libopenssl-conf - 3.0.8-8
libopenssl-devcrypto - 3.0.8-8
libopenssl3 - 3.0.8-8
openssl-util - 3.0.8-8
openvpn-openssl - 2.5.8-1
`

pesa1234, 2023-04-22:
@jbsky I'm trying your commit... I let you know! Thanks again for your hard job

trinidude4, 2023-04-23:
I'm testing that fix too and it looks good for me so far.

olimpiumarius, 2023-04-23:
> > OpenWrt SNAPSHOT, r22599-1416b9bbe9 root@OpenWrt:~# opkg install /tmp/*.ipk Package kmod-mwlwifi (5.15.107+2023-03-11-c916ac13-1) installed in root is up to date. Package mwlwifi-firmware-88w8864 (2023-04-21-ebf38570-7) installed in root is up to date. Collected errors:
> > ```
> > * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.107-1-65a719                   adf77accc904bee4e1a7b5152e) for kmod-mwlwifi
> > ```
> 
> Ops... I compile by myself... So, you can not use (different version I think...) I give to you my Image... 5.15.108 with ksmbd - openvpn -wireguard if you want... just for test...
> 
> Here it is....
> 
> https://drive.google.com/drive/folders/1F0cACS18YonBG4mfbArusU4PgZ3SXM_0?usp=sharing

Hello, 
@pesa1234 is there a possibility to give me the driver again to test it, now I also run Kernel Version 5.15.108!

Thx

pesa1234, 2023-04-23:
Now I have found one more issue... :(

```
Sun Apr 23 17:43:15 2023 kern.err kernel: [  661.699837] rcu: INFO: rcu_sched self-detected stall on CPU
Sun Apr 23 17:43:15 2023 kern.err kernel: [  661.705441] rcu: 	0-....: (1 GPs behind) idle=245/1/0x40000002 softirq=46748/46749 fqs=3000
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.713920] 	(t=6000 jiffies g=50069 q=109)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.718122] NMI backtrace for cpu 0
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.721623] CPU: 0 PID: 54 Comm: kworker/u4:1 Not tainted 5.15.108 #0
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.728093] Hardware name: Marvell Armada 380/385 (Device Tree)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.734036] Workqueue: phy1 ieee80211_ba_session_work [mac80211]
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.740181] [<c010db74>] (unwind_backtrace) from [<c0109a74>] (show_stack+0x10/0x14)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.747965] [<c0109a74>] (show_stack) from [<c04d4c20>] (dump_stack_lvl+0x40/0x4c)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.755574] [<c04d4c20>] (dump_stack_lvl) from [<c04dc554>] (nmi_cpu_backtrace+0xdc/0x110)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.763880] [<c04dc554>] (nmi_cpu_backtrace) from [<c04dc680>] (nmi_trigger_cpumask_backtrace+0xf8/0x130)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.773491] [<c04dc680>] (nmi_trigger_cpumask_backtrace) from [<c018c7c8>] (rcu_dump_cpu_stacks+0x124/0x154)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.783370] [<c018c7c8>] (rcu_dump_cpu_stacks) from [<c018d0a4>] (rcu_sched_clock_irq+0x730/0x8c8)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.792371] [<c018d0a4>] (rcu_sched_clock_irq) from [<c0193948>] (update_process_times+0x88/0xbc)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.801287] [<c0193948>] (update_process_times) from [<c01a4668>] (tick_sched_timer+0x78/0x2b8)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.810026] [<c01a4668>] (tick_sched_timer) from [<c0194560>] (__hrtimer_run_queues+0x158/0x20c)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.818851] [<c0194560>] (__hrtimer_run_queues) from [<c0194e7c>] (hrtimer_interrupt+0x114/0x278)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.827764] [<c0194e7c>] (hrtimer_interrupt) from [<c010d1d8>] (twd_handler+0x34/0x3c)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.835718] [<c010d1d8>] (twd_handler) from [<c017cc84>] (handle_percpu_devid_irq+0x78/0x13c)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.844286] [<c017cc84>] (handle_percpu_devid_irq) from [<c01771f0>] (handle_domain_irq+0x5c/0x78)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.853285] [<c01771f0>] (handle_domain_irq) from [<c04ed0f0>] (gic_handle_irq+0x7c/0x90)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.861501] [<c04ed0f0>] (gic_handle_irq) from [<c0100b7c>] (__irq_svc+0x5c/0x78)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.869018] Exception stack(0xc15dbdc8 to 0xc15dbe10)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.874091] bdc0:                   c3e06000 00000000 000014e1 000014e0 c3ef5480 00000000
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.882304] bde0: c3e061c0 00000000 c3e06000 c28f07bc 00000001 00000000 c3e05fc0 c15dbe18
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.890515] be00: bf236cf0 c0895538 80000013 ffffffff
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.895586] [<c0100b7c>] (__irq_svc) from [<c0895538>] (_raw_spin_lock_bh+0x44/0x58)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.903366] [<c0895538>] (_raw_spin_lock_bh) from [<bf236cf0>] (pcie_tx_xmit+0x168/0x79c [mwlwifi])
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.912492] [<bf236cf0>] (pcie_tx_xmit [mwlwifi]) from [<bf1bb578>] (ieee80211_handle_wake_tx_queue+0x7c/0xc0 [mac80211])
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.923594] [<bf1bb578>] (ieee80211_handle_wake_tx_queue [mac80211]) from [<bf191a3c>] (ieee80211_refresh_tx_agg_session_timer+0xfc/0x3a4 [mac80211])
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.937183] [<bf191a3c>] (ieee80211_refresh_tx_agg_session_timer [mac80211]) from [<bf1929ec>] (ieee80211_stop_tx_ba_cb+0xa4/0xe4 [mac80211])
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.950073] [<bf1929ec>] (ieee80211_stop_tx_ba_cb [mac80211]) from [<bf191550>] (ieee80211_ba_session_work+0x218/0x330 [mac80211])
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.962002] [<bf191550>] (ieee80211_ba_session_work [mac80211]) from [<c01428cc>] (process_one_work+0x210/0x478)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.972295] [<c01428cc>] (process_one_work) from [<c0143448>] (worker_thread+0x4c/0x598)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.980423] [<c0143448>] (worker_thread) from [<c0149e20>] (kthread+0x148/0x164)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.987855] [<c0149e20>] (kthread) from [<c0100130>] (ret_from_fork+0x14/0x24)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  661.995109] Exception stack(0xc15dbfb0 to 0xc15dbff8)
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  662.000181] bfa0:                                     00000000 00000000 00000000 00000000
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  662.008393] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Sun Apr 23 17:43:15 2023 kern.warn kernel: [  662.016605] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000

```

This happen when I ping my router from my smartphone. App used to ping router PingTools.

Previous firmware OK

jbsky, 2023-04-23:
Thank you for your feedback.

Can you reproduce this problem every time with the pingtool application? Did you get a BA session stop in the logs before the kernel error? Do you have a way to check if this log appears?
To force the log to appear on the screen, double the message in the code with wiphy_info and wiphy_warn.

Has anyone else been able to reproduce it? I just tested, I don't reproduce with the pingtools application.

The only thing I see is the following message.
```
ieee80211 phy0: Mac80211 start BA
```
As if the driver never ends a BA session. :)

I came across a similar case, and I think the cleaned buffer doesn't match the stream buffer. I suggest you test this patch, https://github.com/kaloz/mwlwifi/commit/4c2b9db67d2942150846c84ae4253a49f3474ca2.
Attention, 88W8897 is not supported in this patch.

jbsky, 2023-04-23:
[image]
With this commit hum

else spin_lock

hum :(

pesa1234, 2023-04-23:
> [image] With this commit hum
> 
> else spin_lock
> 
> hum :(

Yep, sorry!

closed 2023-04-29 by pesa1234

tamer-hassan, 2023-05-02:
Tested on WRT3200ACM

Compared to the old mwlwifi (without @jbsky patches), the performance is noticeably worse, regardless of whether AMSDU on or off.

i.e. now getting at best 390-400Mbps with iperf3, despite being connected at 1300Mbps (Macbook pro 2019)

Previously, was getting consistent 570-600Mbps (also while connected at 1300Mbps from Macbook pro 2019).

sad :frowning: